ut-robotics/picr22-team-gibibot #10: [Mechanics] How are the ball guides going to be fixed in place

maidlam, 2022-09-21:
How are the ball guides going to be fixed in place so they don't move around when in use?

[image]

marcanderetti, 2022-09-21:
The sticks will be replaced by M6 bolts

Akustav, 2022-09-24:
You need 70mm M6 bolts to make them work. I hope you have those, since I dont think we have such bolts in the lab and they seem quite exotic in their usecases.

marcanderetti, 2022-09-26:
> You need 70mm M6 bolts to make them work. I hope you have those, since I dont think we have such bolts in the lab and they seem quite exotic in their usecases.

I can take bolts from home and cut them to size. not going to be a problem

closed 2022-09-26 by marcanderetti