ut-robotics/picr22-team-gibibot #3: [Mechanics] Not clear where thrower motor cables run

maidlam, 2022-09-21:
From the current model it is unclear from where motor power cables run. Would be great if they can be found on the model, at least a representation of the cable exit area is needed.
Verify the location on your actual motor.

[image]

marcanderetti, 2022-09-22:
motor cables can turn in a 90 degree angle and won't get in the way. Alse we're not quite sure yet where to run the cables

Akustav, 2022-09-24:
I guess the whole motor attachment can be rotated to adjust the cable exit location. Overall I would suggest to run the cable from this area so it does not interfere with the ball or other elements of the thrower.

[image]

closed 2022-09-24 by Akustav