ut-robotics/picr22-team-gibibot #11: [Mechanics] Difficult order of assembly

Akustav, 2022-09-24:
Currently the order of assembly for the motor spacer and motor are a little suspect.

First you need to attach the motor, else the sidewalls block the mounting holes:
[image]

After that you need you logically should attach the motor-spacer assembly to the sidewall but that is made difficult, because the attachment bolt heads can now interfere with the motor body:
[image]

Is this intentional?

marcanderetti, 2022-09-26:
made holes in motor mount so the motor screws could be added later.

marcanderetti, 2022-10-09:
I don't know how many times i have said this, but this is not meant to be adjustable. it just happens to be

Akustav, 2022-10-12:
Okay, I completely misunderstood it then. Can happen with text chats. Overall the design is quite confusing due to so many bolt holes in the component. You could remove the outer bolt circle entirely and mount the motor using its mounting holes, clamping the spacer between the motor and sidewall (blue holes). Otherwise you can remove around half the outer circle holes because there is no use for them (red crosses). The outer circle bolts still interfere with the motor so you can not remove the spacer without removing the motor completely.

A bolt does not in itself stop the pieces from moving relative to each other. The bolt provides clamping force to create friction between the two components. Shear forces apply in this case so a huge number of bolts is major overkill.

[image]

marcanderetti, 2022-10-17:
fixed it

closed 2022-10-19 by Akustav