ut-robotics/picr22-team-gibibot #12: [Mechanics] Side swing stick questions

Akustav, 2022-09-24:
I was wondering how do you plan to manufacture those?

They seem to be a complex solution for a simple issue. Would it not be simpler to use a bolt with a blind threaded hole?

[image]

This reduces the count of unique components while providing the same result

marcanderetti, 2022-09-26:
I was planning to 3d print the stick, and screw from the other side. I thought it would look cooler like this.

Akustav, 2022-10-01:
What advantage except being cool would that provide? It seems more fragile and uses unique components that when lost/broken are more difficult to replace.

Additionally, how are these pieces fastened? The pieces themselves are 6mm, so what kind of screw will hold them in place?
[image]
The same issue is with the other side component that has the fastening hole on the swing arm.

Another thing is the weird sizing of the component stopper sides. One side has a 1mm lip, the other side has 6.4mm
[image]
I think you mentioned earlier that you had to add a hole to make room for the motors shaft. Why have the piece so long at all?

marcanderetti, 2022-10-02:
if the 6.4mm piece is so long, it will be held in place by the motor and won't need fastening. The other side will have a bolt and a spacer

Akustav, 2022-10-03:
Can't we just remove the side swing arms and all components associated with them? The main benefit of these swing arms is smooth motion when adjusting the throwing angle in runtime. This design does not support this approach.

You plan to clamp the internal moving components with the bolt. The two bolts have already fixed spacing with a grey piece. So the bolts can move in the slot and you can clamp the component sandwich without the arms.
[image]

Akustav, 2022-10-12:
I guess it can add a little support. Still one question remains that how will the ramp pivots fasten to the sidewalls.

Hole 1 is 3mm, hole 2 is 2.5mm. What combination of bolts and other fasteners are used there?

[image]

marcanderetti, 2022-10-19:
made both 2.5mm inside, so a screw can be added later. Also changed the direction of the swing stick, so a screw could be added from the outside
[image]

closed 2022-10-20 by ReikoR