ut-robotics/picr22-team-gibibot #13: [Mechanics] Upper guide stick and spacer question

Akustav, 2022-09-24:
Was wondering out of what you plan to do the spacer and stick?
[image]

Would it not be simpler to use a 10mm female-female spacer and 2 flat screws?

marcanderetti, 2022-09-26:
doesn't really matter, I was planning on 3d printing and press-fitting the stick. The model currently does not allow tolerance for screws

Akustav, 2022-10-01:
I do not think 3D-printing is a appropriate solution. It is not a catch-all answer. It should be attempted to avoid bespoke components that are ultra-specific. Maybe there is a better solution like using a F-F spacer fastened with recessed screw.

marcanderetti, 2022-10-02:
well then i can use a threaded rod and nuts. Really anything works that fits in between. It's not like it's going to mass production

Akustav, 2022-10-03:
In most cases it is good to use off-the-shelf components because they are easy to acquire and replace. It is good practice cause it reduces development time. Making one custom component is easy but when you need to 20 of said components, taking them off a shelf is better use of time and effort.

Anyways... Press-fitting 3D-printed sticks may break or take a lot of time to properly fit because tolerances for them are not that great. The stick may also be different in diameter across its length.

Threaded rods + nuts may interfere with the sliding action when you plan to fix it on both sides

marcanderetti, 2022-10-03:
i was meaning to cut a threaded rod to be the length of the stick. But anyways, by our availability of components, 3d printing a 5-minute print 4 times is still going to be faster than going to the hardware store. also, these thing don't have to be press-fitted THAT hard 😄

Akustav, 2022-10-03:
Compared to a hardware store it may be a good solution yeah but we should have a collection of spacers and these would be instantly available. I would also suggest using countersunk screws instead of normal socket cap head screws. It looks sleeker and does not interfere with any other moving parts.

marcanderetti, 2022-10-09:
Well i can look into it while manufacturing

Akustav, 2022-10-19:
It is possible to resolve before "looking into it while manufacturing". I suggest using 10 mm female-female spacers with M3 countersinks. This fixes everything in place rigidly.

marcanderetti, 2022-10-19:
Changed everything to adapt a 10mm spacer

marcanderetti, 2022-10-19:
[image]

closed 2022-10-20 by ReikoR