ut-robotics/picr22-team-gibibot #14: [Mechanics] too many blue lines

Akustav, 2022-09-24:
Currently sketches have too many blue undefined lines. All lines should be black and fully defined. This is important because when changes are required, all sketch dimensions are pre-determined and funky stuff can not happen. If lines are blue, any dimension change can have unforeseen consequences to the component.

Some examples:

[image]
[image]
[image]

marcanderetti, 2022-09-26:
Everything should be fixed now

Akustav, 2022-10-01:
I think everything is not fixed. Some things I found:

[image]
[image]

Overall I also think you have way to many sketches per component so it is hard to understand at times what is in scope for the component.

marcanderetti, 2022-10-02:
Will it really be a problem?

Akustav, 2022-10-03:
Yes. When I moved them so did the slots. For future components any wrong drag or sketch change can create a cascade of weird sketch transformations.

closed 2022-10-12 by Akustav