ut-robotics/picr22-team-gibibot #15: [Mechanics] thrower motor too high

Akustav, 2022-10-01:
The motor seems too high compared to ball height. The ball might not touch the motor correctly

marcanderetti, 2022-10-02:
Awesome. The motor height was determined by instructors in the classroom

Akustav, 2022-10-03:
@maidlam @ReikoR WDYT? 
[image]

maidlam, 2022-10-04:
Me and @ReikoR added these lines, one of them signifying the ground (on the right in the image, other line is on the bottom of the plate), into the assembly during a practical. I can't recall, if it accounted for sinking into the carpet.
The distance to the center of the motor from the bottom line is 55,115 mm, so 41,115 mm should be from the bottom of the motor to the ground.
Since a rubber tube or other soft material is planned to be added to the thrower motor, I think it will add enough thickness, so a ball can be grabbed.

[image]

Akustav, 2022-10-12:
Okay. With the rubber pay mind not to use too soft material that can deform in with high motor rotation speeds.

closed 2022-10-12 by Akustav