ut-robotics/picr22-team-gibibot #36: Ball sensor

ReikoR, 2022-11-05:
[image]

Is the servo used for angle adjustment or ball holding?

For holding a ball sensor should be added. Otherwise it's hard to know when the servo should be stopped.

In the lab we have IS471FE sensors and 3mm IR LEDs.

raimo0, 2022-11-05:
The servo was for angle adjustment but I added a second servo for ball holding and a connector for the IR sensor.

ReikoR, 2022-11-06:
[image]
[image]

PA4 pin is not 5V tolerant.
From [datasheet](https://www.st.com/resource/en/datasheet/stm32g441kb.pdf):
[image]
[image]

Also `SERVO_PWN` should probably be `SERVO_PWM`:
[image]

raimo0, 2022-11-06:
Changed the IR sensor pin to PF0 and fixed the typos.

closed 2022-11-06 by ReikoR