ut-robotics/picr22-team-gibibot #45: [Mechanics] had to assemble top plate

Akustav, 2022-11-16:
If you want to have back support for the motors you should make the fit for these tabs loose to reduce the headache of assembly. The more tight fitting components there are, the harder the robot is to assemble.

[image]

marcanderetti, 2022-11-20:
Made motor mount ends diagonal so the assembly would be easier
[image]
[image]

Akustav, 2022-11-30:
Overall I would suggest having room on both X and Y axis

marcanderetti, 2022-12-02:
I will manualli sand down the edges after manufacturing

closed 2022-12-03 by Akustav