ut-robotics/picr22-team-gibibot #52: [Mechanics] no power plug or switches

Akustav, 2022-11-16:
There are currently no places for a power plug or switches on the chassis

marcanderetti, 2022-11-21:
Made holes for switches, power connectors etc
[image]

Akustav, 2022-11-30:
Current holes would not make it possible to mount switches, for example. I suggest consulting with some guidelines from here:
https://ut-robotics.github.io/picr22-home/basketball_robot_guide/mechanics/design.html#_pc_power_adapter_socket

marcanderetti, 2022-12-01:
Made adjustments 
[image]
[image]

marcanderetti, 2022-12-02:
[image]

closed 2022-12-03 by Akustav