ut-robotics/picr22-team-gibibot #66: No electronics attached to the robot

Akustav, 2022-11-30:
You should have a place for the mainboard and step-up converter inside the chassis.

marcanderetti, 2022-12-01:
[image]
[image]

Akustav, 2022-12-03:
You will most likely not be able to fit the SUB in the mainboard without some additional cutouts:

[image]

marcanderetti, 2022-12-03:
Made a bigger hole in the camera holder, so cables could be routed
[image]
[image]

closed 2022-12-03 by Akustav