ut-robotics/picr22-team-gibibot #68: No room for screw that holds extended ramp to robot

Akustav, 2022-12-03:
There seems to be no room for a normal screw here:

[image]

Akustav, 2022-12-03:
I also suggest to use countersunk screws here as well:
[image]

marcanderetti, 2022-12-03:
I will be using countershrunk screws in places like these
[image]

marcanderetti, 2022-12-03:
Also made the front a bit thicker, for added strength
[image]

Akustav, 2022-12-03:
The component is 3D printed so you should model the chamfers there as well. Drilling them afterward can compromise the 3D printed component.

marcanderetti, 2022-12-04:
added chamfter on the parts
[image]
[image]
[image]

closed 2022-12-04 by Akustav