ut-robotics/picr22-team-gibibot #69: Thrower ca

Akustav, 2022-12-03:


marcanderetti, 2022-12-03:
I deemed it too much to sketch, so i just shortened the cable
[image]

Akustav, 2022-12-03:
I think you missed the point. Where does the cable run? The hole is above the top plate so the cable cant go into the chassis.

[image]

marcanderetti, 2022-12-04:
Oh right... I completely missed that little detail 🤣 
Made a little hole thingy for the cable
[image]

closed 2022-12-04 by Akustav